ps2/rileylink #6: app: feature to configure pump as mysentry broadcast (pair)

ps2, 2015-07-23:
MySentry has the user put the pump into a discovery mode, where it broadcasts packets including the pump id.  The MySentry sends a response with its own id, which the pump then adds to its list of devices it accepts queries from. The pump also then starts broadcasting pump status messages every five minutes.

The screen should walk the user through this process, using a paired RileyLink, and display status updates as it receives packets from the pump and verifies it can talk to the pump.

loudnate, 2015-07-23:
Have you captured the packet that needs to be sent back?

ps2, 2015-07-23:
I believe the packets involved have these types:

https://github.com/ps2/minimed_rf/blob/master/lib/minimed_rf/messages/message_type_map.rb#L10-L11

I'll see if I can find full examples in my packet log.

ps2, 2015-07-23:
It looks like this:

a2 AAAAAA 0a BB CCCCCC DD EE

AAAAAA = pump id
BB = ? (my example had 8f)
CCCCCC = mysentry id (mine was 000695)
DD = ? (my example had 01)
EE = crc

loudnate, 2015-07-30:
Will the RileyLink capture the pump's broadcast packets during discovery mode? I lack a lot of understanding in how the actual IC Rx/Tx flows work, but I'm eager to figure this out.

ps2, 2015-07-30:
Yes, the Rileylink will listen for the pump's message, since it has the pump ID in it, and that way the user doesn't have to enter it in.

loudnate, 2015-08-10:
Have you started this feature? I can take a stab at it this week since I'm going to have to go through this process anyway with my rig.

ps2, 2015-08-10:
I haven't started it quite yet; I am working on the packet sending code, so this particular area is in a bit of flux. I think I'm close, though.  Feel free to start if you want; just be warned that the sending code in RileyLinkBLEDevice is changing.

loudnate, 2015-08-12:
What's the significance of "0a" in the MySentry discovery confirmation packet?

Here's the packet format sent by the pump in "Find Device" mode:
`a2 594040 09 BB 999999 00 EE`

BB increments, usually as a single counter, which keeps the CRC (EE) changing, but the byte prior is always 09. Does 09 signify a discovery while 0a signifies a confirmation?

ps2, 2015-08-12:
That's my take on it as well: https://github.com/ps2/minimed_rf/blob/master/lib/minimed_rf/messages/message_type_map.rb#L10-L11

loudnate, 2015-08-12:
I played with this for a bit but didn't have any luck pairing yet. Any idea on the frequency and repetitiveness of the link messages?

The `BB` sequence in the pump message is a counter that increments by 1, though it sometimes repeats in the same second. It's not time-based, though: if you exit the Find Device mode and start over, it will pick up where it left off. I wonder if it needs to be replayed in the link message?

loudnate, 2015-08-14:
From @ps2: https://gist.github.com/ps2/f80ff7a400bcecfd9479

loudnate, 2015-08-14:
Here's what worked for me:
```objc
@interface MySentryPairingViewController () {
    unsigned char _messageCounter;
}

- (void)viewDidLoad {
    [super viewDidLoad];
    [self.device setTXChannel:3];
    _messageCounter = 0;
}

- (NSString *)mySentryReplyForPacket:(MinimedPacket *)p
{
    static NSString *MySentryID = @"000888";

    NSString *packetStr = [@"a2" stringByAppendingFormat:@"%@%02x%02x%@00%02x000000",
                           [[Config sharedInstance] pumpID],
                           MESSAGE_TYPE_PUMP_STATUS_ACK,
                           messageCounter++,
                           MySentryID,
                           p.messageType];

    return packetStr;
}
```

It's possible to replay multiple messageTypes in a single `06` packet, and @ps2's paring log shows this, but I simply replied immediately during my spoofing.

PR #18 (2015-08-15): MySentry pairing setup screen
Fixes #6
closed 2015-08-15 by ps2