ps2/rileylink #42: 722 2.4A Set Clock fails

edengh, 2016-07-12:
Tried setting clock with RileyLink on a Medtronic 722 Firmware Version 2.4A.
error result:
```
UnexpectedResponse(MinimedKit.PumpMessage(packetType: MinimedKit.PacketType.Carelink, address: <725960>, messageType: MinimedKit.MessageType.EmptyHistoryPage, messageBody: MinimedKit.UnknownMessageBody(rxData: <0c>)), MinimedKit.PumpMessage(packetType: MinimedKit.PacketType.Carelink, address: <725960>, messageType: MinimedKit.MessageType.ChangeTime, messageBody: MinimedKit.ChangeTimeCarelinkMessageBody))
```
Set clock also failed through mmeowlink / OpenAPS TI stick.
I haven't tried this directly with Carelink stick yet.

mhaeberli, 2016-07-12:
I submitted this (accidentally submitted under the wrong userid - edengh - our daughter vs myself)...

loudnate, 2016-07-12:
Is your pump in 24-hour time? Can you include the raw data from the read_clock command?

mhaeberli, 2016-07-12:
pump is on "12 hour" time.
I'll include read_clock later today when I'm near pump again.

mhaeberli, 2016-07-12:
```
openaps use pump read_clock
"2016-07-12T16:16:30"
```

mhaeberli, 2016-07-12:
for what it's worth - I now confirmed that set_clock works within mmeowlink / OpenAPS

I'll try to run a regression test vs RileyLink

loudnate, 2016-07-12:
Any idea what changed between the point where you said it wasn't working with decocare and when you said it was?

mhaeberli, 2016-07-13:
Nate:
Thanks for all your help ; sorry for the distraction.
No idea what changed.
In the 'regression' below, I was using mmeowlink / TI Stick / OpenAPS, until the very end, when I used RileyLink Change Time.
By the way, note the change in the pump time from Zulu to Pacific... (23... to 16...).
This time, RileyLink reported 'succeeded'; it may have been a range issue (that is, nearness of pump to RileyLink or phone to RileyLink).
```
... $ openaps use pump read_clock
"2016-07-12T23:31:03"
... $ openaps use pump set_clock --to now
{
  "raw": [
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0
  ], 
  "requested": {
    "old": "2016-07-12T23:31:14", 
    "params": [
      23, 
      33, 
      41, 
      7, 
      224, 
      7, 
      12
    ], 
    "clock": "2016-07-12T23:33:41.462020"
  }, 
  "timestamp": "2016-07-12T23:33:42.303133", 
  "clock": "2016-07-12T23:33:42"
}
... $ openaps use pump set_clock --to 2016-07-12T23:30:55
{
  "raw": [
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0, 
    0
  ], 
  "requested": {
    "old": "2016-07-12T23:41:25", 
    "params": [
      23, 
      30, 
      55, 
      7, 
      224, 
      7, 
      12
    ], 
    "clock": "2016-07-12T23:30:55"
  }, 
  "timestamp": "2016-07-12T23:41:26.037818", 
  "clock": "2016-07-12T23:30:55"
}
... $ openaps use pump read_clock
"2016-07-12T23:31:08"
... $ openaps use pump read_clock
"2016-07-12T16:42:10"
... $
```
Thanks,
Martin

loudnate, 2016-07-13:
Glad it's working. Thanks!

loudnate, 2016-07-13:
@ps2 I can't close this issue (thought we were on the iOS repo), but it's working as intended.

closed 2016-07-13 by ps2