ps2/rileylink #45: Provide battery voltage

trixing, 2016-11-14:
It would be great to have a way to ask rileylink for the voltage of the internal battery to alert users when it needs to be charged.  Not sure if that's possible with the current hardware or would need some changes?

trixing, 2017-01-13:
Most notable  P0_7 - P0_0 are connected to the ADC, so a voltage divider of VBAT to any of these would do the trick.

ps2, 2017-01-13:
Yes, I've looked at doing a vdiv, but decided that would be constantly draining power (though I haven't done the work to figure out how low I can make that leak and still measure accurately).

So I designed a version that used a mosfet to turn on and off a vdiv measuring circuit, but I had issues with my prototypes (must've screwed up something). At that point I decided I'd sunk too much energy into the issue. 

For me, the main goal here is to minimize the effort users have to spend on maintaining charge on the RL.  Nightly charging is the simplest method, in my opinion.  I'd rather have a habit that I don't think about than have to look at battery status routinely. I'd even take a nightly charge over a 4 or five day charge.  

Back when android phones had way worse battery life than iPhones (they've both moved in directions that make them more comparable now), I had an android and my wife had an iPhone.  I was forced into nightly charging, whereas she could go several days w/o charge.  Guess who had more annoying impacts due to running out of power at inconvenient times?

So, this issue has been thought through (maybe too much).  As usual, I'd love PRs that are a result of tested prototypes.  Otherwise this kind of report is usually just idle speculation from a back-seat designer. ;)

closed 2017-01-13 by ps2

trixing, 2017-01-13:
Typically you would use a 10k / 10k Ohm divisor, which would put current at 3.7V / 20000 = 0.185mA.  Given that this is a 850mAh battery it would only affect the battery life very minimally.

Do you actually know what is drawing so much power to drain the 850 mAh that quickly?

ps2, 2017-01-13:
Yes, I think my original idea to save power with a mosfet was probably overkill, but after spending all that time on it, and realizing that it wouldn't make much of a difference in my use of the device, I decided not to spend more time chasing after it. Again, feel free to work on an updated design, test it, and submit a PR.  As in software, idle comments on how things can be improved without actual contribution can be irritating to the people that work through all the details of actual implementation.  In hardware it's even worse, as the cycles are so long. 

Regarding your second comment, are you referring to the regulator issue?  If so, please move that comment over there.