ps2/rileylink #47: The voltage converter design should follow recommendations from the datasheet

trixing, 2017-01-13:
http://www.ti.com/lit/ds/symlink/tps63031.pdf #11.2

in particular the placement of the capacitors could be better.

Also I wonder how the induction coil was chosen?  It seems to be 7x the internal resistance than the recommended types in the datasheet (0.7 Ohm vs. 0.1 Ohm or less).

(I have a problem with the tps chip getting really hot, but still keeping voltage - I'm theorizing there is some stray capacitance/inductance with some bad feedback. Everything else works.)

ps2, 2017-01-13:
I've gone through several versions of the design that follow the datasheet layout and component selection more closely, and have still been having issues.

Check out the https://github.com/ps2/rileylink/tree/v1.0t branch, for the latest.

ps2, 2017-01-13:
Actually, the v1.0t is using a completely different regulator.  Check out v0.8test2 for previous versions that use the same regulator but follow the datasheet very strictly.

trixing, 2017-01-13:
I'm surprised that still causes issues - what kind of issues if I might ask?  Did you consider just using a 3.3V LDO?

ps2, 2017-01-13:
Same issues of overheating, drawing 200mA.  Yes, looked at 3.3V ldo; wanted a more efficient design, so the new one is a buck.

trixing, 2017-01-26:
I just tried this.  I have a Rileylink which pulls 300mA from the Lipo constantly.  To test if the regulator is broken I directly connecting 3.3V _after_ the regulator.  This lead to a power draw of 0.02A and a working communication with the pump.

Next step I put in a LM1117-3.3V (bad choice but the only one I had around, drop-out of 1V, thus 3.1V at the CC1110).  With a voltage of 5V applied at the input side my current draw is also only 0.02A.  The efficiency seems to be similar and the Rileylink is working again.

I have to get a LDO with less drop-out voltage, otherwise I can only get around 20-30% capacity out of the battery.  Luckily there is a bunch of alternatives and I just ordered an LP2992 which supposedly only has around 0.1V drop voltage.

(As background I can recommend reading

http://electronics.stackexchange.com/questions/38782/regulated-3-3v-from-a-lithium-ion-or-lipo-battery )

closed 2017-03-10 by ps2