ps2/rileylink #49: Power failure

jessicakarma, 2017-03-10:
Hi, I've using RileyLink and Loop since 1/20/2017.  In the past few days Loop has not been connecting (indicated by red circle) and I realized that RileyLink is not holding a charge.  Starting today it doesn't appear to be working at all, even when plugged in.

I saw there is an update coming.  Question:  Do I need to buy another one, or will a replacement be sent automatically?  If I buy another, of course I'd like to have the new version.

Thanks!

ps2, 2017-03-10:
The green light is not a power light. It's a bluetooth connection light. Have you tried diagnosing your bluetooth connection?  Try restarting the Loop app, and bluetooth on the phone, and see if the green light comes on.

Also, check your battery connector to make sure it's seated all the way in (on some rileylinks this can be a very tight fit).

jessicakarma, 2017-03-10:
Hi Pete, thanks much for your quick reply!

I’ve been restarting both the Loop app and the phone; sometimes this works but unfortunately not today.  Other bluetooth devices continue to work including Dexcom G5, so I take it the issue is with the RileyLink connection?

I also opened RileyLink and set the battery and connector i proper place.  Al components are sung and seem to be where intended.

Any other suggestions?

Love Loop, and a big part of that is the streamlined equipment made possible by you!  Many, many thanks for this!! :-)

Take care,
Jessica

ps2, 2017-03-10:
Send me a DM on gitter or facebook, and I'll see if we can get you a replacement.

closed 2017-03-10 by ps2