ps2/rileylink #57: Parts list

javaboyjunior, 2019-04-24:
Hi there

4 of the parts in bom.csv are obsolete on Digikey.com. I have been in touch with Digikey and they have provided part numbers for replacement products. Will you accept a pull request for an updated bom.csv and confirm the specs of the 4 new parts if I make the edit?

Thanks

Mike

xs10shl, 2019-04-30:
Second the request. I'm getting the same Digi-Key errors, and would like to piggyback on the homework already done by javaboyjunior.  

Also, If there's an alternate part number for the Omnipod antenna, perhaps that could be added as well. I see a few choices for the 433mhz antenna, none of which look exactly ghe same as the version you used.  ANT-433-HESM-ND appears to be the closest.

javaboyjunior, 2019-04-30:
Here's my list including replacement parts as recommended by Digikey Tech Support.

[bomupdatedmike.csv.zip](https://github.com/ps2/rileylink/files/3130036/bomupdatedmike.csv.zip)

xs10shl, 2019-04-30:
Thx much, I'll put eyes on it tomorrow as well.

stephenclong, 2019-05-19:
The parts list looks like it's specific to the Medtronic version?  Can someone who knows way more about this than me provide details of which parts I need to switch out to make it work for the Omnipod?  Is it the same PCB?

javaboyjunior, 2019-05-19:
I asked in the support channel and was told the Parts are the same other
than the antenna. I used this for the antenna...

DAOKI 10pcs 433MHz antenna...
https://www.amazon.com/dp/B01CGMOYYM?ref=ppx_pop_mob_ap_share

ps2, 2019-05-19:
Yes the parts on the 433 MHz RL right now are the same as 915 MHz.  But we use this antenna: https://www.ebay.com/itm/10Pcs-433MHz-Helical-antenna-High-quality-Useful-with-Good-Quality/323119548775

ps2, 2019-05-19:
Looked over your replacements, and they look good @javaboyjunior. Thanks!

closed 2019-05-19 by ps2