ps2/rileylink #64: Antenna 433MHz for omnipod rileylink in parts list

jbayart, 2019-11-26:
For the omnipod rileylink, it is a 433 MHz antenna but there is no branch omnipod where the antenna is replaced in the csv in the hardware folder. I am not sure which antenna is used to build the riley.

breakery, 2019-11-26:
Not sure if I get your question right: still a simple helical antenna is used such as this one https://www.amazon.com/433MHz-Antenna-Control-Intercom-Helical/dp/B06Y691H5T

jbayart, 2019-11-26:
Yes I will probably use one like this. But I would like the exact same antenna like the other antenna with the digikey part number to be sure the dimensions fit with the case.

ps2, 2019-11-26:
There isn't a digikey part for this. These are the same part: https://www.ebay.com/itm/10Pcs-433MHz-Helical-antenna-High-quality-Useful-with-Good-Quality/323119548775

jbayart, 2019-11-26:
great, thanks !

closed 2019-11-26 by jbayart

larsvandoremalen, 2020-02-03:
Just to check, the ebay seller indicated does not ship to NL (almost nowhere in europe) and I read somewhere that the antenna signal quality has a lot of influence on creating screamers, would any of the digikey components below be good enough as well? 

https://www.digikey.nl/product-detail/nl/pulselarsen-antennas/W3127/553-1696-ND/2538463
https://www.digikey.nl/product-detail/nl/linx-technologies-inc/ANT-433-HESM/ANT-433-HESM-ND/413457
https://www.digikey.nl/product-detail/nl/linx-technologies-inc/ANT-433-HETH/ANT-433-HETH-ND/625468

There is also an [aliexpress product](https://www.aliexpress.com/item/32962433768.html?spm=a2g0o.productlist.0.0.d3d9468e1zvlNe&algo_pvid=1957e3db-ab41-43e0-8e0f-a13ba98c685d&algo_expid=1957e3db-ab41-43e0-8e0f-a13ba98c685d-59&btsid=413febaa-89e9-4ad0-bbbd-8a2d05120dae&ws_ab_test=searchweb0_0,searchweb201602_4,searchweb201603_53) with a very low VSWR. But not certain if that makes it any better than the digikey ones. 

First time posting, only just got started on building a loop. Want to do as much by myself as possible to fully understand the system. Thanks a lot for all the hard work that went into that!

breakery, 2020-02-04:
The aliexpress looks fine. Have a look here as well: https://www.amazon.de/dp/B00SO651VU/ref=cm_sw_em_r_mt_dp_U_N8roEb6AJERDF 

Those are simple and tested. Range up to approx. 10m.

larsvandoremalen, 2020-02-05:
Thanks! Afraid even the amazon.de one does not ship to the Netherlands. They normally do but for components it's still a bit of hit and miss. Will try the aliexpress one then, thanks for the help.

breakery, 2020-02-05:
If urgent I can send one from de -> nl.

larsvandoremalen, 2020-02-05:
No urgency at all luckily, only preparing to start looping in March/April when I have everything ready. Good to now buy these components with some spare parts already :) Very nice of you to offer though, thanks!

breakery, 2020-02-05:
otherwise have a look here if you want to have some casing and inductive charging: https://sugarloop.makery.de/

ps2, 2020-02-05:
@breakery We have been testing out various designs of inductive charging, water resistant cases for the [official RileyLink site](https://getrileylink.org/), and we have one that is working *very* well; better than any of the others I've seen. It's going to be injection molded, so it's taking a little while to get the tooling set up.