ps2/rileylink #65: BOM for 433 Mhz version

konovagit, 2020-06-11:
Hi Pete,

I have a question concerning the 433 MHz version. Do you have a bom.csv for this version ? 

OtherwiseI just have to change the antenna and the balun, right? I see this version https://getrileylink.org/product/rileylink433 so if you have the component ref it will be great.

Thank you for your job :)

jaredkzr, 2020-10-13:
I'm curious about this as well. The 433 MHz version purchased from https://getrileylink.org/product/rileylink433 appears to have the 915 MHz balun installed which would cause a reduction in range. I can't confirm this though since we have no BOM for the 433 MHz version.

It does appear that changing the antenna and balun is all that is necessary as far as the hardware is concerned. The CC1110 transceiver will have to be configured for 433 MHz though.

hansdampf01, 2020-10-22:
just found this issue: https://github.com/ps2/rileylink/issues/57 where the same question was asked and Im also facing it.
It seems like the antenna module of the 915mhz version will be need you just replace the actual antenna. e.g. to this:

https://www.amazon.de/gp/product/B07QGQ1VG5/ref=crt_ewc_title_huc_1?ie=UTF8&psc=1&smid=A3BN2T8LLIRB5S